EarthScope/evalresp #73: evalresp handling of open-ended end time representations in RESP

rcasey-iris, 2019-12-20:
Want to find a consistent rule for forward-going RESP files to show an open-ended end time, then have evalresp be capable of handling it.  Better yet, it may be good for evalresp to be lenient to what may reasonably be entered as an 'empty' end time.  Transcript below explains:

	Whereas the old RESP service would show a 2599 end time for open station epochs, the new one is currently displaying null:

###################################################################################
#
B050F03     Station:     ZOMB
B050F16     Network:     AF
B052F03     Location:    ??
B052F04     Channel:     BHZ
B052F22     Start date:  2006,152,00:00:00
B052F23     End date:    null
#
#                  +-----------------------------------+
#                  |    Response (Poles and Zeros)     |
#                  |        AF  ZOMB       BHZ         |
#                  |     06/01/2006 to null      |
#                  +-----------------------------------+
#
B053F03     Transfer function type:                A
B053F04     Stage sequence number:                 1
B053F05     Response in units lookup:              M/S - Velocity in Meters Per Second
B053F06     Response out units lookup:             V - Volts....

	In the olden days, the alternative representation was 'No ending time'.  What do we want the new standard to be in future RESP output?

rcasey-iris, 2019-12-20:
@chad-iris replies:

> I think it should either be 'No ending time', or even better leave it empty like:

> B052F22     Start date:  2006,152,00:00:00
> B052F23     End date:    
> Problem is that evalresp does not work with either of these.  Dang, wish we had fixed that in evalresp 5.  We'll do that in the next release.
> 
> For now how about filling in:
> 
> B052F23     End date:  2599,1
> I tested with both evalresp 5 and 3.3.3.
> 
> Chad

chad-iris, 2020-01-28:
evalresp, current and at least back to 3.3.3, accepts the string "No Ending Time" as an open end time.

closed 2020-01-28 by chad-iris