EarthScope/evalresp #76: RESP File Format recognized ht

ivor52, 2020-05-07:


ivor52, 2020-05-07:
when calling evalresp_filename_to_channels(), specifying (options.station_xml = -1) to auto-detect the file format, old format RESP is marked as XML resulting in the following failure message:

```
mxml: Bare < in element!
Thu May  7 17:02:34 2020 [ERROR] Could not parse input
```
sample failing file attached.

richard

[RESPHGN.txt](https://github.com/iris-edu/evalresp/files/4594463/RESP.HGN.txt)

chad-iris, 2020-05-08:
I think this is the problem: `(options.station_xml = -1) `

Instead, try setting that value to 0.  Somewhere during coding I changed my mind on how to trigger auto-detection of format should work/be triggered.  Will consider how to address this in the next release.

ivor52, 2020-05-09:
i detect the resp file format myself, so i will just specify 0 or 1 for whichever it is.

ivor52, 2020-05-09:
to confirm,```options.station_xml``` values are:
- 0 => RESP format
- 1 => XML format

ivor52, 2020-05-09:
n.b. ```libsrc/evalresp/public_api.h::452``` still indicates that -1 should be used for auto-detect.

and / but:
```libsrc/evalresp/input.c::2756``` converts -1 input to 0.  how would setting to 0 result in a change in behaviour?

chad-iris, 2020-05-09:
> that means there's no way to force RESP format?

I believe so, no way to force RESP format.

> libsrc/evalresp/input.c::2756 converts -1 input to 0. how would setting to 0 result in a change in behaviour?

I'd be interested in your programs' behavior when you set `options.station_xml = 0`.

ivor52, 2020-05-09:
results:
```options.station_xml = 0``` - file is parsed without error and succeeds
```options.station_xml = -1``` - file parsing fails with mxml error msg of original post

ivor52, 2020-05-09:
for the moment, i will set the option to 0 for RESP format and 1 for XML format.

chad-iris, 2021-02-26:
I believe this comes down to two issues:
* make sure the in-code and "manual" documentation are correct and clear on this issue
* no mechanism to force input to be considered RESP

chad-iris, 2021-12-25:
Change in bb5948f23cb79bcd965aea880a0e00eef871a06b fixes the documentation of the option.

Closing this.  If the lack of a mechanism to force files to be considered RESP is needed another ticket can be opened.

closed 2021-12-25 by chad-iris