EarthScope/evalresp #79: Expand List of Recognized Units

timronan, 2020-05-15:
Evalresp does not recognize the unit radian. The list of units that it recognizes should be expanded to include radian, and potentially all of the units that are accepted by the stationxml validator. 
https://github.com/iris-edu/stationxml-validator/wiki/Unit-name-overview-for-IRIS-StationXML-validator 

```
Fri May 15 09:46:57 2020 [WARN] check_units; units found ('RAD - OPTICAL PHASE') are not supported
Fri May 15 09:46:57 2020 [WARN] check_units; units found ('RAD - OPTICAL PHASE') are not supported
Fri May 15 09:46:57 2020 [ERROR] DA.UI1.00.SD0: undefined units
```

metempleton, 2021-08-04:
It also needs to include radians per second.  We need evalresp to handle the attached example before we can release the NRL web service.
[RESP.XX.NS529..BJZ.TELLUS_R2.30.50.txt](https://github.com/iris-edu/evalresp/files/6933879/RESP.XX.NS529.BJZ.TELLUS_R2.30.50.txt)

chad-iris, 2021-12-23:
This also applies to #84.

After reading the source code and thinking through this issue I've come to the conclusion that we do not want to add more recognition of units to evalresp as it continues a bad pattern, will probably never be complete, and is not needed.

Background material:

First, evalresp _can_ work with any units by specifying the "default" units, e.g. the `-u DEF` option.

Second, the default output of evalresp is velocity, specifically m/s, which is the `-u VEL` option.  Providing a response that cannot be converted to m/s results in an error, that makes sense and there are probably use cases that rely on this behavior.  This is historical and probably should not be changed until a major revision (not even on the horizon) if at all.

Third, the "recognized" units of evalresp like PA (PRESSURE), C (CENTIGRADE), T (TESLA), only serve to avoid the failure and _are otherwise equivalent to requesting default units_.  The evalresp program literally does nothing else with them.  The calling program (using the API) might have access to this value, so one could argue there is a small advantage of having evalresp parse the values for you, but it's hardly an advantage as it is very limited and partial broken (PRESSURE is only triggered by "PA", there is no recognition of micro/milli/mega prefixes, etc.).

Summarizing these points: the "recognized", non-convertible units of evalresp are only useful as a "shortcut" to avoid specifying default units as output.  Furthermore, they are inconsistent with a request for DIS, VEL, and ACC in that they will "work" but _not_ be in the requested units; unless something special is done for these other units, the hard-coded exceptions are bad practice.

If my understanding outlined above is correct, I do not think we should add any more unit recognition to evalresp, it serves very little purpose and extends a problematic pattern of usage.  Instead, **programs and users should always request the default units unless they wish to limit their usage to DIS, VEL or ACC (in meters)**.

The iris `tracedsp` program addresses this by making DEF the default, which is used by the `irisws-timeseries` service for signal operations and thus gains the same default.  The `irisws-evalresp` services does the same, specifying units in the response as the default.  These are well behaved, side-stepping the historical quirk of evalresp's default VEL output by making a new default from the start of their existence.

Other major software using evalresp: `sac` defaults to `DIS`.  ObsPy defaults to `VEL` (I think).  In both cases the appropriate result when a response is specified that cannot be converted to those units is to fail.

I've got a change in progress to clarify what the units are in user output, evalresp is not very good about this reporting, in particular when a limited stage range is used.

Also, I'll try to improve this error:
```
Fri May 15 09:46:57 2020 [WARN] check_units; units found ('RAD - OPTICAL PHASE') are not supported
```
Which should really say that the units cannot be converted to DIS, VEL, or ACC in meters.  That's what it means.  It does not mean they are not supported at all.

metempleton, 2021-12-23:
This would work if -u DEF mean the units listed in the RESP file and not m/s 
(velocity) across the board.  

Evalresp generates a warning when units are “not supported” (that’s ok), but 
JPlotResp and JEvalResp terminate with 

Error parsing response from “RESP_filename":  Invalid in-units type ("CELSIUS - temperature in degrees Celsius") in Coefficients Blockette at line 2

even when the user specifies "-u DEF”.  

—Mary

chad-iris, 2021-12-23:
> This would work if -u DEF mean the units listed in the RESP file and not m/s (velocity) across the board. 

That is the meaning of the option as I understand it and as far as I can tell how evalresp operates.

> Evalresp generates a warning when units are “not supported” (that’s ok), 

Do you have an example?  I ran evalresp 5.0.0 on your provided RESP and there is no warning.  See output below.  I tried with evalresp 3.3.3 also, and there was no warning either.

> but JPlotResp and JEvalResp terminate with Error parsing response from “RESP_filename": Invalid in-units type ("CELSIUS - temperature in degrees Celsius") in Coefficients Blockette at line 2 even when the user specifies "-u DEF”.

In my opinion that means those programs are in error, and inconsistent with evalresp.

Output using `-u DEF` with "unsupported" units:
```
% evalresp NS529 BJZ 2010 1 0.01 100 100 -f RESP.XX.NS529.BJZ.TELLUS_R2.30.50.txt -vv -u DEF
Thu Dec 23 14:13:01 2021 [INFO] << EVALRESP RESPONSE OUTPUT V5.0.0 >>
Thu Dec 23 14:13:01 2021 [INFO]  --------------------------------------------------
Thu Dec 23 14:13:01 2021 [INFO]   RESP.XX.NS529.BJZ.TELLUS_R2.30.50.txt
Thu Dec 23 14:13:01 2021 [INFO]  --------------------------------------------------
Thu Dec 23 14:13:01 2021 [INFO]   XX NS529 ?? BJZ 
Thu Dec 23 14:13:01 2021 [INFO]  2006,001,00:00:00.0000 No Ending Time
   Seed units: rad/s - Rotational velocity in radians per second(in)->COUNTS - Digital Counts(out)
Thu Dec 23 14:13:01 2021 [INFO]    computed sens=4.98686E+02 (reported=5.00000E+02) @ 1.00000E-01 Hz
Thu Dec 23 14:13:01 2021 [INFO]    calc_del=0.00000E+00  corr_app=0.00000E+00  est_delay=0.00000E+00  final_sint=0.025(sec/sample)
Thu Dec 23 14:13:01 2021 [INFO]       stage  1: LAPLACE     A0=1.843958E+08 NZeros=  4 NPoles=  7 Sd=4.986862E+02
Thu Dec 23 14:13:01 2021 [INFO]       stage  2: FIR_SYM_1   H0=1.000000E+00 Ncoeff=  1 SamInt=2.500000E-02 Sd=1.000000E+00
Thu Dec 23 14:13:01 2021 [INFO] --------------------------------------------------

```

closed 2022-09-28 by chad-iris