EarthScope/evalresp #81: ResponseList in StationXML is ignored

chad-iris, 2021-02-09:
The attached StationXML file contains a single stage response described as a Frequency-Amplitude-Phase listing.  evalresp 5.0.0 does not recognize this `<ResponseList>` declaration.

The stand-alone `xml2resp` converter also fails to output anything related to the `<ResponseList>`.

[IM.VNDA..BHE.xml.gz](https://github.com/iris-edu/evalresp/files/5953357/IM.VNDA.BHE.xml.gz)

chad-iris, 2021-02-09:
Conversion of the response containing a `<ResponseList>` with `xml2resp`, note the lack of a response list (FAP):

```
% xml2resp IM.VNDA..BHE.xml 
Tue Feb  9 09:41:21 2021 [INFO] Logging to stderr
Tue Feb  9 09:41:21 2021 [INFO] Input from IM.VNDA..BHE.xml
Tue Feb  9 09:41:21 2021 [INFO] Output to stdout
#
###################################################################################
#
B050F03     Station:     VNDA
B050F16     Network:     IM
B052F03     Location:    ??
B052F04     Channel:     BHE
B052F22     Start date:  2000,031,00:00:00
B052F23     End date:    No Ending Time
#
#                  +-----------------------------------+
#                  |           Response List           |
#                  |        IM  VNDA       BHE         |
#                  |     01/31/2000 to No Ending Time      |
#                  +-----------------------------------+
#
B055F03     Stage sequence number:                 1
B055F04     Response in units lookup:              M - EARTH DISPLACEMENT IN METERS
B055F05     Response out units lookup:             COUNTS - DIGITAL COUNTS
B055F06     Number of responses listed:            0
#
#                  +-----------------------------------+
#                  |            Decimation             |
#                  |        IM  VNDA       BHE         |
#                  |     01/31/2000 to No Ending Time      |
#                  +-----------------------------------+
#
B057F03     Stage sequence number:                  1
B057F04     Input sample rate (HZ):                 2.0000E+01
B057F05     Decimation factor:                      00001
B057F06     Decimation offset:                      00000
B057F07     Estimated delay (seconds):             +0.0000E+00
B057F08     Correction applied (seconds):          +0.0000E+00
#
#                  +-----------------------------------+
#                  |      Channel Sensitivity/Gain     |
#                  |        IM  VNDA       BHE         |
#                  |     01/31/2000 to No Ending Time      |
#                  +-----------------------------------+
#
B058F03     Stage sequence number:                 1
B058F04     Sensitivity:                           +1.33000E+07
B058F05     Frequency of sensitivity:              +1.00000E+00
B058F06     Number of calibrations:                0
#
#                  +-----------------------------------+
#                  |      Channel Sensitivity/Gain     |
#                  |        IM  VNDA       BHE         |
#                  |     01/31/2000 to No Ending Time      |
#                  +-----------------------------------+
#
B058F03     Stage sequence number:                 0
B058F04     Sensitivity:                           +1.33000E+07
B058F05     Frequency of sensitivity:              +1.00000E+00
B058F06     Number of calibrations:                0

```

chad-earthscope, 2026-11-07:
Fixed in 2a5eec3 and 94d74e1

closed 2026-11-07 by chad-earthscope