EarthScope/evalresp #83: Evalresp doesn't parse startDate of 1970-01-01T00:00:00 when reading FDSNstationXML

metempleton, 2021-07-28:
When reading stationXML, evalresp doesn't meaningfully parse the date 1970-01-01T00:00:00.  (However, it does 
successfully parse this date for RESP and it can parse 1970-01-01T00:00:01 for stationXML).  The attached test file 
generates the following output:

Wed Jul 28 14:05:49 2021 [INFO] << EVALRESP RESPONSE OUTPUT V5.0.0 >>
Wed Jul 28 14:05:49 2021 [INFO]  --------------------------------------------------
Wed Jul 28 14:05:49 2021 [INFO]   IU.ANMO.00.BHZ.20210701_time.xml
Wed Jul 28 14:05:49 2021 [INFO]  --------------------------------------------------
Wed Jul 28 14:05:49 2021 [INFO]   IU ANMO 00 BHZ 
Wed Jul 28 14:05:49 2021 [INFO]  No Ending Time No Ending Time

with call
evalresp ANMO BHZ 2021 182 0.001 5000 5001 -f IU.ANMO.00.BHZ.20210701_time.xml.txt -vx

[IU.ANMO.00.BHZ.20210701_time.xml.txt](https://github.com/iris-edu/evalresp/files/6896201/IU.ANMO.00.BHZ.20210701_time.xml.txt)

chad-iris, 2021-07-28:
I believe the problem is rooted here:
https://github.com/iris-edu/evalresp/blob/master/libsrc/evalresp/stationxml2resp/xml_to_dom.c#L230 

(which is ironically documented as "separated out for testing", but appears to being used in production).

The result for 1979-01-01 is converted to a value of 0.

This is used here:
https://github.com/iris-edu/evalresp/blob/master/libsrc/evalresp/stationxml2resp/xml_to_dom.c#L266

and then here:
https://github.com/iris-edu/evalresp/blob/master/libsrc/evalresp/stationxml2resp/xml_to_dom.c#L1094

where the return value of 0 means the "failure" pretty sure.

chad-iris, 2021-12-24:
The time value of `1970-01-01T00:00:00` is properly converted and status of the functions is returned as "OK".  The problem is this test of the epoch value which assumes 0 is means OPEN:

https://github.com/iris-edu/evalresp/blob/17ab910f023b5c273a1e91dada65d66cf565a662/libsrc/evalresp/stationxml2resp/dom_to_seed.c#L69

The code assumes that the only OPEN value is an end time and uses the open end time message of "No Ending Time", and there is no equivalent for start time.

closed 2021-12-24 by chad-iris

chad-iris, 2021-12-24:
The commit fixes the recognition of a date-time of 1970-01-01T00:00:00.  Test file now results in:

```
Thu Dec 23 22:07:47 2021 [INFO]  --------------------------------------------------
Thu Dec 23 22:07:47 2021 [INFO]   IU.ANMO.00.BHZ.20210701_time.xml
Thu Dec 23 22:07:47 2021 [INFO]  --------------------------------------------------
Thu Dec 23 22:07:47 2021 [INFO]   IU ANMO 00 BHZ 
Thu Dec 23 22:07:47 2021 [INFO]  1970,001,00:00:00 No Ending Time
...
```

The lack of a reasonable open time message for a start date is moot because the startDate attribute in StationXML is still required by the code.  The startDate attribute is not required by the 1.1 schema, so this remains as incomplete support of StationXML 1.1.

commit 2ca00102 (2021-12-24): Handle 1970-01-01T00:00:00 values in StationXML. Fixes #83