EarthScope/evalresp #85: EvalResp requires the optional StationXML Poles:Number Zeros:Number Attribute

timronan, 2021-12-10:
Evalresp throws an error if it encounters a stationxml poles and zeros stage where the Poles:Number and or Zeros:Number attributes are missing. These are optional attributes, and are not required by the stationxml schema. Below is an example of a Poles/Zeros cascade that are missing the number attributes.   

```
              <Zero>
                <Real>0.0</Real>
                <Imaginary>0.0</Imaginary>
              </Zero>
              <Zero>
                <Real>0.0</Real>
                <Imaginary>0.0</Imaginary>
              </Zero>
              ...
              <Pole>
                <Real>-0.036614</Real>
                <Imaginary>0.037059</Imaginary>
              </Pole>
              <Pole>
                <Real>-0.036614</Real>
                <Imaginary>-0.037059</Imaginary>
              </Pole>
              <Pole>
                <Real>-32.55</Real>
                <Imaginary>0.0</Imaginary>
              </Pole>
            ...
```

Evalresp should be able to function without this optional attribute.  Below is an example of a Pole/Zeros cascade that will work with the current of version of evaresp. 
```
              <Zero number="0">
                <Real>0.0</Real>
                <Imaginary>0.0</Imaginary>
              </Zero>
              <Zero number="1">
                <Real>0.0</Real>
                <Imaginary>0.0</Imaginary>
              </Zero>
              ...
              <Pole number="0">
                <Real>-0.036614</Real>
                <Imaginary>0.037059</Imaginary>
              </Pole>
              <Pole number="1">
                <Real>-0.036614</Real>
                <Imaginary>-0.037059</Imaginary>
              </Pole>
              <Pole number="2">
                <Real>-32.55</Real>
                <Imaginary>0.0</Imaginary>
              </Pole>
            ...
```

chad-iris, 2021-12-10:
A response-level StationXML file for testing.
[GO_TRLT_00_HHZ.xml.gz](https://github.com/iris-edu/evalresp/files/7694620/GO_TRLT_00_HHZ.xml.gz)

chad-iris, 2021-12-12:
Fixed in commit ea171441087c3ae47ad3d1e854fa958992e6efd1

closed 2021-12-12 by chad-iris