EarthScope/evalresp #86: Fault reading StationXML

chad-earthscope, 2023-04-06:
The input StationXML, also attached:
```
% curl -o IU_MAJO_LDO.xml 'https://service.iris.edu/fdsnws/station/1/query?net=IU&sta=MAJO&chan=LDO&level=resp&format=xml&nodata=404'
```
On macOS this fault is produced:
```
% evalresp MAJO LDO 2022 015 0.01 1.0 -f IU_MAJO_LDO.xml      
Thu Apr  6 13:55:35 2023 [WARN] IU.MAJO.30.LDO: Missing gain in stage 1 - using unit gain at 1Hz
Thu Apr  6 13:55:35 2023 [WARN] IU.MAJO.31.LDO: Missing gain in stage 1 - using unit gain at 1Hz
Thu Apr  6 13:55:35 2023 [ERROR] Cannot compute B62 response for negative or zero input: 0.000000
evalresp(34182,0x7ff8466e5340) malloc: *** error for object 0x600002b68040: pointer being freed was not allocated
evalresp(34182,0x7ff8466e5340) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      evalresp MAJO LDO 2022 015 0.01 1.0 -f IU_MAJO_LDO.xml
```

[IU_MAJO_LDO.xml.gz](https://github.com/EarthScope/evalresp/files/11174368/IU_MAJO_LDO.xml.gz)

metempleton, 2023-05-26:
In addition to getting this error on stationXML, I'm also getting it with RESP:

met% evalresp CMB LKS 2005 001 0.001 5000 5001 -f RESP.BK.CMB..LKS -b62_x 50 -u 'def' -v
Fri May 26 08:06:25 2023 [INFO] << EVALRESP RESPONSE OUTPUT V5.0.1 >>
Fri May 26 08:06:25 2023 [WARN] BK.CMB..LKS: Missing gain in stage 1 - using unit gain at 1Hz
Fri May 26 08:06:25 2023 [WARN] BK.CMB..LKS: Missing gain in stage 9 - using unit gain at 1Hz
Fri May 26 08:06:25 2023 [INFO]  --------------------------------------------------
Fri May 26 08:06:25 2023 [INFO]   RESP.BK.CMB..LKS
Fri May 26 08:06:25 2023 [INFO]  --------------------------------------------------
Fri May 26 08:06:25 2023 [INFO]   BK CMB ?? LKS 
Fri May 26 08:06:25 2023 [INFO]   2004,167,00:00:00.0000 2010,351,00:00:00.0000
Fri May 26 08:06:25 2023 [INFO]    documented input units: DEGC - degree Celsius
Fri May 26 08:06:25 2023 [INFO]    documented output units:  ÔíÔfÝ
Fri May 26 08:06:25 2023 [INFO]    requested units: Documented response unit
Fri May 26 08:06:25 2023 [INFO]    computed sens=4.23160E+05 (reported=0.00000E+00) @ 0.00000E+00 Hz
Fri May 26 08:06:25 2023 [INFO]    calc_del=1.58233E+01  corr_app=1.59062E+01  est_delay=1.58110E+01  final_sint=1(sec/sample)
Fri May 26 08:06:25 2023 [INFO]       stage  1: POLYNOMIAL     MacLaurin Coeffs= 10
Fri May 26 08:06:25 2023 [INFO]       stage  2: FIR_SYM_2   H0=1.000000E+00 Ncoeff=  0 SamInt=1.953125E-04 Sd=4.137690E+05
Fri May 26 08:06:25 2023 [INFO]       stage  3: FIR_SYM_2   H0=1.000000E+00 Ncoeff= 64 SamInt=1.953125E-04 Sd=1.014770E+00
Fri May 26 08:06:25 2023 [INFO]       stage  4: FIR_SYM_2   H0=1.000000E+00 Ncoeff= 72 SamInt=3.125000E-03 Sd=9.781120E-01
Fri May 26 08:06:25 2023 [INFO]       stage  5: FIR_SYM_2   H0=1.000000E+00 Ncoeff= 64 SamInt=1.250000E-02 Sd=1.011130E+00
Fri May 26 08:06:25 2023 [INFO]       stage  6: FIR_SYM_2   H0=1.000000E+00 Ncoeff= 64 SamInt=2.500000E-02 Sd=1.011130E+00
Fri May 26 08:06:25 2023 [INFO]       stage  7: FIR_SYM_2   H0=1.000000E+00 Ncoeff= 64 SamInt=5.000000E-02 Sd=1.011130E+00
Fri May 26 08:06:25 2023 [INFO]       stage  8: FIR_SYM_2   H0=1.000000E+00 Ncoeff=260 SamInt=1.000000E-01 Sd=9.967120E-01
Fri May 26 08:06:25 2023 [INFO] --------------------------------------------------
evalresp(2379,0x7ff8494a7340) malloc: *** error for object 0x600000f8d420: pointer being freed was not allocated
evalresp(2379,0x7ff8494a7340) malloc: *** set a breakpoint in malloc_error_break to debug
Abort

Interesting that the output units (COUNTS for stage 0) are not being relayed correctly... 

[RESP.BK.CMB..LKS.txt](https://github.com/EarthScope/evalresp/files/11575972/RESP.BK.CMB.LKS.txt)

chad-earthscope, 2026-11-07:
Fixed in v5.1.0 with 6c13e89

closed 2026-11-07 by chad-earthscope