EarthScope/evalresp #88: evalresp 5.0.1 returns "[ERROR] Could not parse input" for polynomial sensor + linear datalogger response

metempleton, 2024-06-10:
I downloaded the RESP file for BK.CMB..LKS for 2004,167,00:00:00 to 2010,351,00:00:00 using the resp ws, but this command:

% evalresp BK CMB 2005 001 0.001 5000 5001 -b62_x 1 -u 'def' -vx -f RESP.BK.CMB..LKS.txt
Mon Jun 10 14:12:13 2024 [INFO] << EVALRESP RESPONSE OUTPUT V5.0.1 >>
Mon Jun 10 14:12:13 2024 [ERROR] Could not parse input

could not read it.  This SNCL/epoch is the SEED manual example for how to combine polynomial and linear response stages in the same RESP file.  I've included that RESP file here.  I discovered this issue when trying to troubleshoot a similar response from a user.

[RESP.BK.CMB..LKS.txt](https://github.com/user-attachments/files/15778769/RESP.BK.CMB.LKS.txt)

chad-earthscope, 2024-06-10:
The good news is that the command line includes the `-x` option to force input to be considered to be StationXML, which is not in this case.

The bad news is that removing that option and trying again results in a program fault:
```Console
evalresp BK CMB 2005 001 0.001 5000 5001 -b62_x 1 -u 'def' -f RESP.BK.CMB.LKS.txt 

evalresp(67344,0x7ff84ef4efc0) malloc: *** error for object 0x600003d65200: pointer being freed was not allocated
evalresp(67344,0x7ff84ef4efc0) malloc: *** set a breakpoint in malloc_error_break to debug
```

:(

chad-earthscope, 2024-06-10:
I downloaded the response in StationXML and it results in the same fault.  That XML is attached.

[BK_CMB__L_K_S.xml.gz](https://github.com/user-attachments/files/15779939/BK_CMB__L_K_S.xml.gz)

chad-earthscope, 2026-11-07:
Fixed in v5.1.0 refactoring, with user improvements in dfdef7c511552597b1e60fcee514d01bb1d266b9

closed 2026-11-07 by chad-earthscope